fellesverkstedet/Bark-beetle-parametric-toolpaths #2: Inch conversion slows down calculating

Siemenc, 2018-02-03:
In postprocessor inch conversion slows down the definition - Remove this feature?
[Link to component](https://github.com/fellesverkstedet/Bark-beetle-parametric-toolpaths/blob/master/Bark%20beetle%20components/CNC%20milling%20PostProcessor.ghuser)

Siemenc, 2018-02-03:
[image]

Siemenc, 2018-02-23:
I wanted to fix this but I'm not sure which one is the latest update on the postprocessor component: See #43

JensDyvik, 2018-02-25:
Fixed with new version of the CNC milling post processor. The inch conversion option has been removed. Much safer anyway since the inch conversion was only a test and not reliably implemented.

Very interspersing that a simple dividing by 1 operation could create so much calculation time.

closed 2018-02-25 by JensDyvik