fellesverkstedet/Bark-beetle-parametric-toolpaths #18: pockets offset overshoot (copied from file)

Siemenc, 2018-02-06:
Bug copied from file

Siemenc, 2018-02-09:
Example file for this?

Siemenc, 2018-02-09:
Is it for examples like this? Because I found the error that causes this and fixed it. As soon as I got my github up and running I'll commit the updated version

[image]

Siemenc, 2018-02-09:
Github up and running. Committed latest pocket component which solves this.

closed 2018-02-09 by Siemenc

JensDyvik, 2018-02-25:
Awesome to have this fixed. Thank you very much. The fix has been included in the main CNC milling example in the latest development folder: https://github.com/fellesverkstedet/Bark-beetle-parametric-toolpaths/commit/b1dd65f60118ba676e5350a7cde896db4f392525