fellesverkstedet/Bark-beetle-parametric-toolpaths #23: fix cut inside inside inside (copied from file)

Siemenc, 2018-02-06:
Bug copied from file

Siemenc, 2018-02-06:
I think I solved this for both cutout and pocket. I copied a piece of code and did some fixing on making the trees match. I've only tested on the example below (gray uses old code, green is using edited code). In the pocket component this also seems to solve some unwanted artifacts happening in this specific case (see red marked areas).
[image]
I think I'll first do a bit more testing before replacing the pocket and cutout component and committing.

Siemenc, 2018-02-08:
I started doing some testing and found out I also had to do a fix in the case of using only 1 curve. Now it seems to be working quite well. In the case of the pocket it seems like it's also way faster.
[image]

Siemenc, 2018-02-09:
Tried committing with Github. But I get an error saying "Failed to sync this branch - You might need to open a shell and debug the state of the repo."

Siemenc, 2018-02-09:
Seems like it worked now. I completely disabled my antivirus.

closed 2018-02-09 by Siemenc

JensDyvik, 2018-02-25:
Very exciting to have this fixed. Thank you very much. The fix has been included in the main CNC milling example in the latest development folder: https://github.com/fellesverkstedet/Bark-beetle-parametric-toolpaths/commit/b1dd65f60118ba676e5350a7cde896db4f392525

I dont have a specific torture testing docuement. Maybe you could upload the one you made to a test object folder in the repo?

JensDyvik, 2018-02-25:
By the way, as mentioned before, the planar surface function is a potential method for solving endless objects inside objects. And extracting isocurves from the planar surfaces could potentially be a fast and reliable way of creating pocket raster lines.

Siemenc, 2018-02-26:
I added a folder with the curves I've been using for testing. Some caused issues before and are still causing issues (the H with the snake right now for example). I'll also keep using this as a place to store new objects that cause issues while I encounter them.

Yeah the planar surface method for solving endless objects inside objects is a really good idea. It requires a bit more data tree matching than the solution I integrated now. I think for now this works but it's definitely something to look into for the future!