fellesverkstedet/Bark-beetle-parametric-toolpaths #25: fix pocket raster (copied from file)

Siemenc, 2018-02-06:
Bug copied from file

Siemenc, 2018-02-07:
Is this related to the problem that was solved in [this issue](https://github.com/fellesverkstedet/Bark-beetle-parametric-toolpaths/issues/23)?

Siemenc, 2018-02-12:
I think I've fixed the pocket raster, if this is the problem?
[image]

Siemenc, 2018-02-23:
As soon as I committed the new pocket component I'll be closing this. There's still some room for improvement on the offset rastering but as shown in issues #41 and #39 most of it has been improved.

closed 2018-02-23 by Siemenc

JensDyvik, 2018-02-25:
Awesome awesome. The bug description was aimed at general improvement of the mastering logic, which it seems you have contributed a lot to. Is now included in the latest example file