fellesverkstedet/Bark-beetle-parametric-toolpaths #39: Pocket with curves inside that form an island get recognized as a curve inside a curve

Siemenc, 2018-02-12:
See example below: Area with X should create a pocket as well. But in current logic it defines it as an area inside an area and therefore skipping it.
[image]

Siemenc, 2018-02-16:
Here's another example of issues with curves inside inside. This works well: 
[image]
Now if I add these curves on the bottom right, it messes up the inside inside.
[image]

Siemenc, 2018-02-22:
With sorting out the offset component as described in #41  I also fixed these issues. Very satisfying to see these special cases being calculated. Even a tiny bit faster than the old offset component[image]

closed 2018-02-22 by Siemenc

JensDyvik, 2018-02-25:
Greeeeeeeeeeeeeeat!   As mentioned in the other issues, the improved pocket function is now included in the latest CNC milling example. https://github.com/fellesverkstedet/Bark-beetle-parametric-toolpaths/commit/b1dd65f60118ba676e5350a7cde896db4f392525