fellesverkstedet/Bark-beetle-parametric-toolpaths #49: Potential speed enhancement: an easier+quicker align curves in same direction

Siemenc, 2018-11-07:
I came across this solution for making curves go the same direction. Seems to speed up things a lot (x 100 in example below) since there's no need for offset.
[image]

JensDyvik, 2018-11-07:
Exciting. Don't you need to know whether a curve is going in a climb or conventional direction before you decide to flip? I havent found a good way to analyze if a curve is climb or conventional. Also note that curves inside a curve needs to go in the other direction than the outer curve since they are inside cuts

Siemenc, 2018-11-07:
_Don't you need to know whether a curve is going in a climb or conventional direction before you decide to flip?_
Yes, that's what the circle does. Flip curve has an input G which is a guide curve to check against whether the input in C is in the same direction or not. If the direction is different it switches it to match the same direction.

Also just for clarity: It doesn't speed up the whole component by 100x. The solution above just makes a part of the component 100x faster.

_Also note that curves inside a curve needs to go in the other direction than the outer curve since they are inside cuts_
Yep, that's what happens after this step inside the cutout (and other) component . It basically just would replace this:
[image]

JensDyvik, 2018-11-08:
Fantastic. Very nice solution. This combined with creating boundary surfaces from cutout and pocket curves to find curves inside curves will be a great combo

closed 2018-11-08 by JensDyvik