fellesverkstedet/Bark-beetle-parametric-toolpaths #51: Barkbeetle Rhino OSX

Makershub, 2019-03-21:
Just a note that Barkbeetle is working on OSX, however, CNC cut is not being registered so no cut data is going through the system.

This is most likely because of a silly difference between OSX and windows...

Maybe we can look at it in person.

Note, Drill, Pocket, engrave and 3D are all working. Tabs can't be tested due to CNC cut not working.

Siemenc, 2019-04-06:
I bugtracked this on my girlfriend´s laptop and it seems to be actually be related to the settings generator component. 
[image]
It seems to be a bug on Mcneel´s side. For some reason it doesn´t recognise the numbers from the panel below as numbers but as text. But then when I copy the text from that panel and paste into a new panel it does work.
[image]

Siemenc, 2019-04-06:
It gets even more odd. After I pasted it into the panel and closed the cluster, the settings generator works and it generates the machine code for cutting.
[image]

Now, when I open the settings generator cluster again. The same panel that was working before stops working and gives the same output as the original panel was giving.
[image]

JensDyvik, 2019-04-09:
Nice bug hunting Siemen!  Maybe we can fix this by making a file in rhino6 for windows? It is a bit annoying with the small changes and bugs between differnt generations of rhino and grasshopper

Siemenc, 2019-05-10:
I've just seen this post on the rhino forum which seems to describe the
issue we're having.
https://discourse.mcneel.com/t/integers-get-converted-to-strings-when-opening-gh-definition-on-mac/82988

It is supposedly fixed in rhino WIP for mac, meaning also that it will fix
itself as soon as rhino 6 for Mac comes out. Until then or until we manage
to solve this, you could use bark beetle in rhino WIP.

Siemenc, 2019-10-02:
We just did some testing with Rhino 6 and it seems like the file "CNC-milling" from the stable release folder works.  https://github.com/fellesverkstedet/Bark-beetle-parametric-toolpaths/tree/master/Examples%20-%20Stable%20release

But the interesting part is that we for some reason cannot open the latest "CNC-milling - Rhino6" which contains the latest improvements and features. It would be great if somebody else with a mac and Rhino6 could confirm that this cannot be opened from their computer?