fellesverkstedet/Bark-beetle-parametric-toolpaths #58: The Rhino6 .gh file doesn't stream

almowery, 2019-12-18:
I setup Rhino5 to run grasshopper and I have opened the template in Rhino.  Everything looks good in both rhino and grasshopper, but my .gh file does not look like the screen views in the instructions.   There is an error that says there is no connection for the streaming block and I cannot right click and get the popup item to set the streaming directory.  I tried to post the screenshot here.

[image]

Siemenc, 2019-12-18:
I assume setting the streaming directory will not work since the latest grasshopper file has been made using Rhino 6. I see that neither the button for turning streaming on or the path is connected to anything. I think this is because the Python component is included in Grasshopper 1, which is part of Rhino 6.  For using the latest Rhino 5 version, check our archived components: https://github.com/fellesverkstedet/Bark-beetle-parametric-toolpaths/tree/master/Archive/Bark%20beetle%20stable%20release%20Rhino5
You'll need to manually setup the streaming destination though buy zooming out in the grasshopper definition until you see the blue panels on the right with the code. You can set a streaming destination buy right clicking on each panel and choosing a folder. It's a bit cumbersome, which is why I implemented this new solution in Rhino 6 using a python script, but it should get you going. After you set the streaming destination and made sure "streaming" is enabled, you can save the grasshopper file and it will remember the streaming destination for the next time you open the file. Let me know if you get it to work.

almowery, 2019-12-19:
Thank you so much.  I can't believe I did not see the output of the g-code in the blue boxes.  and the streaming directory is set.  How do I enable streaming?

Siemenc, 2019-12-19:
Make sure "Stream contents" is enabled when right clicking on the panel:
https://aws1.discourse-cdn.com/mcneel/uploads/default/original/3X/c/1/c157921f4692c56ef0f135295ac1a14a441bc882.png

Also one thing worth noting: when deciding on a place to stream the file to, you need to choose the name of the file you are streaming. It's important to add the extension of the type of code you want to stream to the end of the name. For shopbot this is ".sbp" at the end of the filename. For G-code this is .gcode.

closed 2019-12-19 by almowery

almowery, 2019-12-19:
I have had success!  What a great tool.  This will do very well.  I was able to make  gcode for 3D very nice.  I will read up on my grasshopper.  I want to make the bits different like less than 1mm tip, etc.  I will contribute this and other mods to the repository.  I will probably move on to Rhino 6 to keep things compatible and useful to others.  Thanks again.