fellesverkstedet/Bark-beetle-parametric-toolpaths #60: Sorting by control points / changing control points

Siemenc, 2020-02-19:
Got some suggestions from somebody on the Rhino forum, which nice explanatory gifs
https://discourse.mcneel.com/t/i-wish-there-was-an-entry-level-cam-solution-for-rhino3d/64135/87?u=siemen

Could be good to have control over where the startpoint of each curve is. Might be for very special cases where it's necessary (like pieces that are positioned close to each other) but would be nice to have for those cases.
Sorting by control points shouldn't be that hard to implement after that.

Siemenc, 2020-05-30:
This has been implemented in the latest commit.

[image]

[image]

closed 2020-05-30 by Siemenc

JensDyvik, 2020-05-31:
Absolutely awesome with sorting implemented and the menu options!