fellesverkstedet/Bark-beetle-parametric-toolpaths #61: How to generate Gcode?

zohozer, 2020-03-18:
Hi. I just installed BarkBeetle and I am trying to generate some Gcode for a MasterWood machine, but unfortunately I can't find a way to start the generation of the Gcode. I made all the setup as described on the front-page but I have a folder with a bunch of .sbp files that are not containing the actual generated Gcode.

[image]

How I can get a preview of the  toolpaths like in this picture?
[image]

[NewProject.zip](https://github.com/fellesverkstedet/Bark-beetle-parametric-toolpaths/files/4348060/NewProject.zip)

Siemenc, 2020-03-18:
If you zoom out in grasshopper there's a button under Post Processor settings that says make gcode not shopbot code:
[image]
[image]
This will create .NC files. Let me know if this works for you.

zohozer, 2020-03-20:
Unfortunately I can't get the results I need. Now I get the ShopBot code in .nc files.

[image]
[image]

Actually nothing seems to be calculated, only the headers are there without any really Gcode.

zohozer, 2020-03-20:
Also, forgot to mention, I can't get any preview of the tool-paths as shown in the picture from the original post.

Siemenc, 2020-03-20:
Hi @zohozer,

You are actually right. That's a clear bug. But fortunately this is an easy fix which you can do before I have time to fix this in the current verison on github: Zoom out to see the canvas and then zoom in on this part on the right, above the blue panels:
[image]
Here you can switch the output extensions: Link .NC to 0 and .sbp to 1.
[image]
And then in your case you'll need to turn "Make shopbotcode not g-code" and "Include start code" to False again.
I recreated these steps on your part and then I get the code streaming to the correct folder.

[image]

A few extra things to note:
- Your object needs to be positioned below 0 to be able to generate code that makes sense. 
- I see you turned on Trochoidal milling under high speed machining. I suggest leaving that off until you get a bit more familiar with how to work with this definition in combination with the machine you are running this on. (it will also speed up calculations)
[image]
- It seems like there's something not going correct with generating the toolpaths for this surface
[image] I'll file this as a separate issue.

zohozer, 2020-03-20:
Thank you. I used Rhino7 WIP.

Switching back to Rhino6, now I do have the Gcode line visualisation on viewport but the generated files still are missing the Gcode inside. I do only have the header for my CNC machine that I declared on the "Custom start G-code" panel.

=============

%
|1 3DFinish1
|2 0
|3 1
|4 500.000
|5 47.991
|6 200.000
|7 0 0 0 0
|8 0
|9
|10
|11
|12
|13
|14
G0 Z25 ;jog to material height
G92 Z0 ; make top of material Z zero
G1 F3600 Z14
M3 S18000 ; start spindle and set spindle speed
G4 P5 ;wait a few seconds for the spindle to start
M5

zohozer, 2020-03-20:
This is how looks the toolpath on Rhino:

[image]

and this is what I do want to achieve (screenshot from Alphacam):

[image]

In alphacam I can limit the mechanisation to a boundary, in this case the green square lines.

It is possible to add a boundary limit also in Bark-Beetle?