fellesverkstedet/Bark-beetle-parametric-toolpaths #65: file type is sbp while disabled

uomou, 2020-06-05:
Using bb on mac OS X v10.15

Its streaming out to a sbp file, while the option is disabled.
Also I get unnamed.\1 Drill - Bark beetle.sbp

How do I delete the .\?
[mill.\4 Cut - Bark beetle.txt](https://github.com/fellesverkstedet/Bark-beetle-parametric-toolpaths/files/4737626/mill.4.Cut.-.Bark.beetle.txt)
[image]

attached a screenshot and a sbp file with txt extension to check it

Siemenc, 2020-06-05:
When did you download the bark beetle master file? This should be fixed in the latest commit 6 days ago: #63

Siemenc, 2020-06-05:
Also: do you manage to run it on Mac without issues? I've gotten feedback from other people saying they can't open the one of the latest versions on Mac.

uomou, 2020-06-05:
Hi Siemen!
And thanks for your swift reply.
I did download it yesterday, and so far it works fine for me, and I am very happy!
Though with the 5 version I never got it to work.
you mean 3dm file or the gh file?

For a test I will download