fellesverkstedet/Bark-beetle-parametric-toolpaths #67: Added option to set Z zero bellow material + warning and better preview

JensDyvik, 2020-07-16:
I have added four new features the main bark beetle file:

1. In post processor setting you can now choose whether Z zero is above or bellow the material/stock
2. The thickness of the material is now visualized. Instead of having a work area rectangle, we now have a work area "box"
3. The preview box is marked with Z coordinates in top and bottom. This should help reduce confusion, and serve as a visual reminder of whether Z zero is set above or bellow the material
4. I made a crude warning system if any toolpath is set to mill too far into the machine bed. Right now it is set to warn if you cut deeper than 1mm

Being able to set Z zero bellow material is great, because it allows a machine to be set up with maximum limits. For instance with GRBL or ShopBot control you can set that the machine is not allowed to too deep into the bed, or move too far outside work area in X, and Y direction. This can help make a machine much more amateur friendly to use, for instance in fab labs and makerspaces.

Once the file has been tested, we can move it over to the stable release folder. Who has access to a machine and can test? The file is available here: https://github.com/fellesverkstedet/Bark-beetle-parametric-toolpaths/blob/master/Development%20files/Bark%20beetle%20-%20CNC%20milling%20-%20Rhino6.gh

[image]

[image]

[image]

[image]

[image]

Siemenc, 2020-07-17:
Nice!
1. Good feature, seems like you managed to implement it in a quite simple way.
2 +3. Makes sense, I like how it marks Z0 and the depth of the material. Makes it easier to check for depth in front view, as I used to do by drawing a line.
4. Also good idea, but I'd include that maximum depth as a setting into the settings somewhere so you can increase this when 3D milling with a ballnose bit for example. Maybe this message can be centered/scaled according to the toolpaths? Alternatively it's quite easy to implement a pop-up message with this message using ghpython, but when I tried testing that I run into issues with selecting things in the grasshopper window.

I'll try to do some testing later/during the weekend and then we can commit it. I also saw a tree matching bug in the pocketing component which can be included at the same time.

JensDyvik, 2020-07-17:
Awesome Siemen. Thank you for testing. I very much agree that a "max depth before warning" setting should be with the other post processor settings.

It would also be super convenient to have warnings about a job *(or go to home position after a job)* being outside work area in X and Y with an adjustable margin too. Maybe we should make one single bounding box for all toolpaths and get warnings for X,Y and Z from the same cluster/group?

I also agree that scaling and positioning of the warning text should be much better. But I couldn't think of a smart and simple approach, so I made rough hack to get us started :)

PS: I have some thoughts on how we could work with a "machine profile" with all the settings in one file or panel. That would make it smoother for users to upgrade to the latest version of upgrade bark beetle without having to input all the settings every time. I will make some tests and a new issue for that

Siemenc, 2020-07-17:
I added a pop up message when you'll cut in the material and thought it might be good to still show it in the viewport as you had, but just a bit more subtle below the job. What do you think? [image]

I agree about those warnings outside of the cutting area. Besides that, there's a lot more things which we can analyze and give warnings about to the user: f.i. curves failing to offset, pockets failing to be generated, self-intersecting curves, ...

2 bugs dealing with data trees have been fixed, which I created earlier by simplifying things in the cutout & pocket component. The fixes are included in the latest development file.

Regarding upgrading to the latest version of bark beetle. Do you think it would make sense to start numbering our versions, so we have better control over the latest file we're working on and people are using?