fellesverkstedet/Bark-beetle-parametric-toolpaths #68: 3D milling incomplete curves, Rhino 5

almowery, 2020-07-28:
I have been using pockets extensively,  couple fixes for rounding.  But overall great software.  3D milling many problems.  Could someone check on rhino 6?  Thanks.  I would like to help fix bugs if i am pointed to some direction.  I have to modify the bit size to go down to .5mm for my wax cutting.  The pic shows pocket(edge cutting) on left object does not go all the way down.  The middle object the milling goes below the object, both pocket and milling.  Ready to move to rhino 6, BB version 5 is just hopefully buggy and version 6 is better.  Thanks.
[image]

[BBtest 004.zip](https://github.com/fellesverkstedet/Bark-beetle-parametric-toolpaths/files/4991529/BBtest.004.zip)

JensDyvik, 2020-07-31:
Hi. Thank you so much for your feedback! The rhino/gh offset function is not so robust unfortunately, and sometimes give buggy results that we dont handle so well with gh list management. I am on holiday now and have laptop break in order to heal some RSI in my wrist, so I cannot have look at your file now. I will get back to you in a few weeks.

In the meantime i recommend you to set up individual pocket jobs for better control. You can also put flat single surfaces in the 3D layer and get automatic pocket depth. You can also manually edit the areas to be pocketed so no gap is too narrow for the milling bit to pass, giving you one continuous pocket from a surface or curves.

When you put geometry in the 3D layer you get an automatic cutout job next to pocket and 3D rroughing and finishing. You can disable the cutout job in the advanced 3D settings panel if you want.

Siemenc, 2020-08-03:
I'm on holidays myself as well but will have a look as soon as I'm back!

almowery, 2020-10-29:
As a followup I fixed some of the problems by doing an offset to the original object before generating g-code.  Rhino 5 again.  I have looked at using the inside cutting to create molds.  Using a simple square/rectangle solid set -z it seems as though the cutting is below the bottom of the object.  Can anyone verify if this happens in Rhino 6 version?  Please. Thanks.

[image]

Siemenc, 2020-11-02:
Is this the same as the previous file you uploaded here or did you forget to upload a new file?

almowery, 2020-11-09:
Hi Siemen,  I just saw this email.  I did not up load a file. I created a solid 10x40x-10(x,y,z) at the origin.  I used the rhino archive version 5 Bark Beetle, parameter changes>>>>>

Pass depth factor to 0.2.Material to wax.Object on the CNC 3D layerInside cutting YES 

I get three passes first at approx. z = -3.5, second z= -7.0, third z=-10.5.
Thank You Again and Again. Al Mowery

almowery, 2020-11-18:
Here is the Rhino 5 file.  Thanks AL

    On Sunday, November 8, 2020, 6:32:35 PM CST, Alfred Mowery <al.mowery@yahoo.com> wrote:  
 
  Hi Siemen,  I just saw this email.  I did not up load a file. I created a solid 10x40x-10(x,y,z) at the origin.  I used the rhino archive version 5 Bark Beetle, parameter changes>>>>>

Pass depth factor to 0.2.Material to wax.Object on the CNC 3D layerInside cutting YES 

I get three passes first at approx. z = -3.5, second z= -7.0, third z=-10.5.
Thank You Again and Again. Al Mowery

JensDyvik, 2020-11-19:
Hi Al

It looks like your file didn't get uploaded. Maybe you need to attach it in a post here instead of attaching it be email?

I wonder if you might have two problems. A misunderstanding of what "Inside cutting" is supposed to do, and the Z depth error that you describe.

The inside cutting option is for manually setting bark beetle to mill the inside of a curve (for instance to make a window opening in a door). It is not meant for cutting the inside of a polysurface. If you want to make a mold you have to model the mold itself, not the object you are going to cast in the mold. This is often easy to do with "Boolean difference" in rhino. I made a test in rhino 6 with that looks like this. It works quite well, with pockets for the flat areas, and 3D roughing and finishing for the fillet and slanted area: 
[image]

I have uploaded a file as a Rhino 5 file. Could you test it and share a screenshot?

About the wrong Z depth, I wonder if it might be that the older version of Bark Beetle is trying to be smart and take into account that you are working with a ball nose bit, and therefore need cut deeper with the radius of the bit extra in order to cleanly cut out the 3D part and not leave a "lip" in the bottom. Could you test with setting your bit type to upcut or downcut and see if the problem persists?

The 3D milling version of Bark Beetle is the least developed and a little buggy unfortunately. If you want to use it for daily 3D milling you will need to do some manual hacking of you 3D models and bark beetle setting now and then.