fellesverkstedet/Bark-beetle-parametric-toolpaths #69: Easy Gallery to add & show images

Siemenc, 2020-09-15:
As I was using Bark Beetle yesterday to create some legs for a drawer unit I'm making I thought it could be nice to have a place to make it easier to post pictures of things made with Bark Beetle.
[image]
[image]
[image]
The next picture is when I realized I shouldn't have put a finishing pass on the other side. The roughing looks super nice as it's following the shape of the legs nicely.
[image]
[image]

JensDyvik, 2020-09-18:
Nice leg making work!  I agree about a better gallery being needed, the pics we have in the bottom of the readme.md now is not a great solution.

Maybe if we move our documentation to a wiki on this repo, we could have dedicated gallery page on the wiki?

By the way, your legs have by coincidence quite similar shape to a part Thomas Winther made with Bark Beetle back in 2016 :) 
[image]

Siemenc, 2020-09-18:
Having a dedicated gallery page on the wiki would be similar to the readme file I think. Here in issues I can just paste the image and I'm done (I don't know where they get saved though?). With wiki/readme I need to upload the picture to the repository, then find the link to that image and then edit the readme file/wiki and paste in the link. And then it's easy to just skip it and do it maybe another time. 

It is indeed similar to that part from Thomas! The legs came of the cnc super clean, I did no sanding whatsoever and just attached them to the furniture.

Here's another one from today, testing latest development version of bark beetle for a project in the coming week at Fellesverkstedet.

[image]
[image]

Siemenc, 2021-03-25:
A bit of a delayed post but the transition at the top and bottom of this piece has been made using Bark Beetle (Bonus: The dome in the background has also been made using Fellesverkstedet)
[image]

[image]

Siemenc, 2021-03-25:
Another one from the archives: In 2019 these Volvo 240 pieces (1/2 scale) were machined using Bark Beetle. It involved quite some 2-sided 3D milling to make the pieces fit together. I mainly did rough machining on most parts to fit the budget of the project. The client then sanded everything down and gave it a paint job and used it for an animation/video.
[image]
[image]
[image]

JensDyvik, 2021-03-25:
Thank you for sharing Siemen. Amazing work! The Volvo parts really show the positive side for making paths by extracting isocurves from nurbs surfaces.