fellesverkstedet/Bark-beetle-parametric-toolpaths #73: Add key to explain preview colors of different toolpaths

JackFelles, 2020-11-23:
[image]

JackFelles, 2020-11-23:
For clarity: just the colors. No need for maps

JensDyvik, 2020-12-07:
That's a great point Jack, thank you!  I will add it to the to do for documentation. It could be a good opportunity to almost tweak some of the default colors, which are either too similar or hard to distinguish from rhinos default grey background.

JensDyvik, 2021-03-02:
The different colors have confused people quite a bit. I have made a test with having all the different toolpaths the same high contrast color in the latest beta: 
[image]

What do you think? The beta is published here, and will go into stable release when it has been properly tested: https://github.com/fellesverkstedet/Bark-beetle-parametric-toolpaths/blob/master/Development%20files/Bark%20beetle%201.02%20beta%20-%20CNC%20milling%20-%20Rhino6.gh

JackFelles, 2021-03-02:
This seems like a good solution. Its clear that these paths relate to the geometry and their function (pocket/cut etc) is readable.
The main place this becomes a bit confusing is when the roughing and finishing passes are generated because you won't be able to differentiate. Maybe I'm seeing a problem that doesn't exist though as its an area I have the least experience with.

JensDyvik, 2021-03-02:
Great to hear that you have a good feeling for the same colors on all toolpaths too! Next up is better default colors for the layers _(CNC Cut, CNC Pocket, etc).

Regarding what is roughing and what is finishing, I think you are right that it is confusing. I think a good solution is to stop having separate roughing and finishing jobs. There is simply a job called "3D milling", and that's it. If both roughing and finishing is on _(which is default)_, the job will do the roughing first and then the finishing in one go. If a users wants to do only roughing or finishing, this can be done be turning finishing or roughing off in the advanced settings. What do you think?

JackFelles, 2021-03-03:
The logic being that if you want to separate the roughing and finishing then you are ready to dive into the advanced settings. This way basic 3d carving is super simple but if you want to complicate things you can. 

Should this be information in the current settings panel? eg 3D roughing on/off, 3D finishing on/off, or is that something that is understood by the tool paths? The tool paths generated are more complex than simple pocketing or cutting. Maybe these need a bit more explination than the other tool paths. But again, you are speaking to someone who is yet to sucessfully use these features so i don't know if I am qualified to comment.

JensDyvik, 2021-03-05:
I think that is unnecessary to add rough and finish on/off to the simplified control panel. If we add too many options to the simplified control panel, it is no longer a simplified control panel. But it is a bit painful that advanced settings are very overwhelming now and not easy to read and change for those who dont know grasshopper. But that is more of a building an interface with or without grasshopper challenge to solve in the future.

95% of bark beetle users at for instance Fellesverkstedet wouldn't need to know that it is possible to separate roughing and finishing into two separate processes. The 3D geometry gets cut, and thats it. It is quite rare that people use different bits for roughing and finishing, and this can still be achieved with the advanced settings

Right now the main reason why rough and finish are sometimes toggled on and off is because the collision detection takes very long to solve. So it can be a bit annoying to wait for both roughing and finishing to be calculated while working on large objects. This could also be solved with better logic for collision detection.

Lets test with merging roughing and finishing into one operation and keeping the simple control panel simple and gather feedback :)

closed 2021-03-05 by JensDyvik