fellesverkstedet/Bark-beetle-parametric-toolpaths #76: UGS - Pause error caused by strange string output

uomou, 2021-02-05:
Using  Rhino 7 on OSX10.15.7 with the latest BB patch, we expirenced multiple pause errors on UGS.
Looking at the NC file, there are strings ending with -15 and -13 after the Z depht value, always the first string of a block.
This happend only on cut as far as I know right now. Will update here if it isnt the case

I attach the NC file for better understanding:

G1 F3600 Z6
M3 S18000 ; start spindle and set spindle speed
G4 P5 ;wait a few seconds for the spindle to start
G1 F3600 Z6
G1 F12000 X140.886387 Y50.383247
G1 F2100 X140.88638740477 Y50.3832474212997 Z-3.5527136788005E-15
G1 F3000 X156.99445991516 Y50.3832474212998 Z-9.3
G1 F3000 X157.463763310281 Y50.4201823995143 Z-9.3
G1 F3000 X157.921510898285 Y50.5300778724143 Z-9.3
G1 F3000 X158.356431414379 Y50.7102278487346 Z-9.3
G1 F3000 X158.757815672038 Y50.9561964381749 Z-9.3
G1 F3000 X159.11578025872 Y51.2619270777401 Z-9.3
G1 F3000 X159.421510898285 Y51.6198916644223 Z-9.3
G1 F3000 X159.667479487725 Y52.0212759220811 Z-9.3
G1 F3000 X159.847629464046 Y52.4561964381749 Z-9.3
G1 F3000 X159.957524936946 Y52.9139440261791 Z-9.3
G1 F3000 X159.99445991516 Y53.3832474212998 Z-9.3
G1 F3000 X159.99445991516 Y151.383247421299 Z-9.3
G1 F3000 X159.986879229516 Y151.5963827001 Z-9.3
G1 F3000 X109.986879229516 Y853.5963827001 Z-9.3
G1 F3000 X109.921447354816 Y854.041080628199 Z-9.3
G1 F3000 X109.790308652156 Y854.471011108798 Z-9.3
G1 F3000 X109.596407004404 Y854.87652279603 Z-9.3
G1 F3000 X109.344095235913 Y855.248512512238 Z-9.3
G1 F3000 X109.039037397625 Y855.578629601741 Z-9.3
G1 F3000 X108.688081616944 Y855.859463391535 Z-9.3
G1 F3000 X108.29910636673 Y856.084709550667 Z-9.3
G1 F3000 X107.880843604455 Y856.249311613768 Z-9.3
G1 F3000 X107.442682751813 Y856.349574491719 Z-9.3
G1 F3000 X106.99445991516 Y856.383247421299 Z-9.3
G1 F3000 X74.9944599151604 Y856.3832474213 Z-9.3
G1 F3000 X74.5251565200397 Y856.346312443085 Z-9.3
G1 F3000 X74.0674089320355 Y856.236416970185 Z-9.3
G1 F3000 X73.6324884159417 Y856.056266993865 Z-9.3
G1 F3000 X73.2311041582829 Y855.810298404425 Z-9.3
G1 F3000 X72.8731395716007 Y855.504567764859 Z-9.3
G1 F3000 X72.5674089320355 Y855.146603178177 Z-9.3
G1 F3000 X72.3214403425952 Y854.745218920518 Z-9.3
G1 F3000 X72.1412903662749 Y854.310298404425 Z-9.3
G1 F3000 X72.0313948933749 Y853.85255081642 Z-9.3
G1 F3000 X71.9944599151603 Y853.3832474213 Z-9.3
G1 F3000 X71.9944599151601 Y53.3832474212998 Z-9.3
G1 F3000 X72.0313948933747 Y52.9139440261791 Z-9.3
G1 F3000 X72.1412903662747 Y52.4561964381749 Z-9.3
G1 F3000 X72.321440342595 Y52.0212759220811 Z-9.3
G1 F3000 X72.5674089320353 Y51.6198916644223 Z-9.3
G1 F3000 X72.8731395716005 Y51.2619270777401 Z-9.3
G1 F3000 X73.2311041582827 Y50.9561964381749 Z-9.3
G1 F3000 X73.6324884159415 Y50.7102278487346 Z-9.3
G1 F3000 X74.0674089320353 Y50.5300778724143 Z-9.3
G1 F3000 X74.5251565200394 Y50.4201823995143 Z-9.3
G1 F3000 X74.9944599151601 Y50.3832474212998 Z-9.3
G1 F3000 X156.99445991516 Y50.3832474212998 Z-9.3
G1 F3600 Z6
G1 F12000 X124.778315 Y50.383247
G1 F2100 X124.778314894379 Y50.3832474212998 Z-1.24344978758018E-13
G1 F3000 X156.99445991516 Y50.3832474212998 Z-18.6
G1 F3000 X157.463763310281 Y50.4201823995143 Z-18.6
G1 F3000 X157.921510898285 Y50.5300778724143 Z-18.6
G1 F3000 X158.356431414379 Y50.7102278487346 Z-18.6
G1 F3000 X158.757815672038 Y50.9561964381749 Z-18.6
G1 F3000 X159.11578025872 Y51.2619270777401 Z-18.6
G1 F3000 X159.421510898285 Y51.6198916644223 Z-18.6
G1 F3000 X159.667479487725 Y52.0212759220811 Z-18.6
[... 2219 lines cut ...]
G1 F3000 X1184.61293027808 Y1527.86646329649 Z-18.6
G1 F3000 X1184.11897045825 Y1529.1098346189 Z-18.6
G1 F3000 X1183.55355072744 Y1530.32238114461 Z-18.6
G1 F3000 X1182.91858428704 Y1531.5 Z-18.6
G1 F3000 X1182.2162196625 Y1532.63870649563 Z-18.6
G1 F3000 X1181.44883343337 Y1533.73464760916 Z-18.6
G1 F3000 X1180.61902219174 Y1534.78411502279 Z-18.6
G1 F3000 X1179.72959375618 Y1535.78355767098 Z-18.6
G1 F3000 X1178.78355767098 Y1536.72959375618 Z-18.6
G1 F3000 X1177.78411502279 Y1537.61902219174 Z-18.6
G1 F3000 X1176.73464760916 Y1538.44883343337 Z-18.6
G1 F3000 X1175.63870649563 Y1539.2162196625 Z-18.6
G1 F3000 X1174.5 Y1539.91858428704 Z-18.6
G1 F3000 X1173.32238114461 Y1540.55355072744 Z-18.6
G1 F3000 X1172.1098346189 Y1541.11897045825 Z-18.6
G1 F3000 X1170.86646329649 Y1541.61293027808 Z-18.6
G1 F3000 X1169.59647435236 Y1542.03375878326 Z-18.6
G1 F3000 X1168.30416502708 Y1542.38003202334 Z-18.6
G1 F3000 X1166.99390808634 Y1542.65057831928 Z-18.6
G1 F3000 X1165.67013702488 Y1542.84448222806 Z-18.6
G1 F3000 X1164.33733106494 Y1542.96108764024 Z-18.6
G1 F3000 X1163 Y1543 Z-18.6
G1 F3600 Z6
G1 F12000 X300 Y600
M5

JensDyvik, 2021-02-05:
Thank you for the bug report Alex. The problem seems to be that Grasshopper has calculated a very long number for the Z value on some positions, and Grasshopper uses by default scientific notation for long numbers *(which uses the E- or E+)*. More info on that here [https://www.grasshopper3d.com/forum/topics/formatting-numbers-in-grasshopper](https://www.grasshopper3d.com/forum/topics/formatting-numbers-in-grasshopper).

I will make an update to the post processor that forces it to use only regular numbers.

GRBL only uses three decimal points anyway, for instance Z-3.553. So i will try to add an option to select number of decimal points too. This will make the gcode files smaller and faster to transfer.

I will get back to you when I have uploaded an attempt to fix the bug.

uomou, 2021-02-05:
Sounds great, but I think it could make sence to round off on 2 numbers only behind the coma.
Everything can then stil be machined with a precision of +- 0,01mm, which is enough I suppose

JensDyvik, 2021-02-05:
Small update. I was able to reproduce the scientific format problem in Rhino6 SR33on Windows10 too. So I think it is related to a recent update to Grasshopper:
[image]

About only two decimal points, it tugs on my precision heart strings :) I think I will set the default to three decimal points, but give you the option to set it to two. After all, modern computers and networks are quite capable of handling a few megabytes

uomou, 2021-02-05:
Fantastic!

JensDyvik, 2021-02-05:
I have uploaded a fix to the "Development files" folder [https://github.com/fellesverkstedet/Bark-beetle-parametric-toolpaths/blob/master/Development%20files/Bark%20beetle%201.01%20-%20CNC%20milling%20-%20Rhino6.gh](https://github.com/fellesverkstedet/Bark-beetle-parametric-toolpaths/blob/master/Development%20files/Bark%20beetle%201.01%20-%20CNC%20milling%20-%20Rhino6.gh). Can you dowload, test and report back?

It is looking good so far: 
[image]

If you double click the "CNC milling PostProcessor" cluster you can access the decimal places panel. I set it to three :)
[image]

JensDyvik, 2021-02-05:
I found the info needed on an easy way to control decimal places in grasshopper here for those interested [https://discourse.mcneel.com/t/how-to-format-numbers-text-to-2-decimal-places/60086/10](https://discourse.mcneel.com/t/how-to-format-numbers-text-to-2-decimal-places/60086/10)

uomou, 2021-02-05:
Great Jens, we will download and test it and report
On Win10 & OSX

uomou, 2021-02-05:
We've been testing on Mac, and it seems to work 
[image]

JensDyvik, 2021-02-05:
Super! I am marking this as solved then.

I made the a small tweak to the settings generator that was maybe not the smartest timing. Before it would turn on ramping if you worked with a downcut, ballnose or end mill in Hard wood type material. I turned that off because ramping is almost never needed in hard wood, and often made trouble in 3D milling. So now you only get ramping with compression bits and any type of bit when working in plastic, alu or steel.

So if you have set your material type to "Hard wood" that would explain why dont see a z-3,552... coordinate with the new version, because it comes from ramping around a corner. If you want to double check that everything is ok, also when ramping into the material you can tell barkbeetle that you are working with "Plastic" or manually turn on ramping

closed 2021-02-05 by JensDyvik