fellesverkstedet/Bark-beetle-parametric-toolpaths #77: List matching bug in 3D mill

Siemenc, 2021-03-12:
Was getting some issues with milling a quite easy surface today, 1 of the 2 problems I found was that something was off with list matching causing unnecessary jogging and less efficient toolpaths. It was an easy fix once I found the problem. I'll upload version 1.01, which fixes this, later.
Before:
[image]

After
[image]

Siemenc, 2021-03-12:
Looks like I was behind on the latest version. :-) Will upload it as 1.02 Beta 2, as discussed with @JensDyvik

closed 2021-03-12 by Siemenc

JensDyvik, 2021-03-13:
Awesome. Thank you for fixing Siemen! I was not aware of this list management error. We had so many list management bugs from gh in rhino5 to gh in rhino6. Luckily it seems that list management has not changed in rhino7.