fellesverkstedet/Bark-beetle-parametric-toolpaths #79: future bark beetle feature request

uomou, 2021-05-31:
complexer structures require multible jobs like pocketing, engraving and cutout without a tool change.
In this case, it would be nice to be able to render all those part jobs into a buffer, and bake 1 including gcode file.
This would allow one to just start up a multi job with engravings pockets and cutouts, do something else while it's milling,
and get to assembly when job is done.

off course its relaitvely quick to do this manually, and thats what I do,
but it makes bark beetle a little bit more a Maserati milling software:-)

JensDyvik, 2021-06-01:
Thanks for the feedback Alex. I agree with you and have some thoughts already for how this could be done nicely in bark beetle 2.0.

For now it is possible rewire the grasshopper definition so that all calculated toolpaths go into the same post processor and then the same file. Could be worth testing. But be carefull to not run a screw hole hold down job twice!

uomou, 2021-06-01:
Thats sound sgreat Jens.
For the time being, I have mostly the same multi job, being the bird house,
So I stay careful and do it by hand. But great that it is possible to do!

Hilsen,

Alex

From: Jens Dyvik
Sent: Tuesday, June 1, 2021 4:09 PM
To: fellesverkstedet/Bark-beetle-parametric-toolpaths
Cc: uomou; Author
Subject: Re: [fellesverkstedet/Bark-beetle-parametric-toolpaths] future barkbeetle feature request (#79)

Thanks for the feedback Alex. I agree with you and have some thoughts already for how this could be done nicely in bark beetle 2.0.
For now it is possible rewire the grasshopper definition so that all calculated toolpaths go into the same post processor and then the same file. Could be worth testing. But be carefull to not run a screw hole hold down job twice!