fellesverkstedet/Bark-beetle-parametric-toolpaths #80: another Future version request

uomou, 2021-07-02:
For multi tasking people like me running a maker space.

It would be terrific if the NC file could have a header line above G1 F3600 Z6
mentionn the milling bit diameter and type, say like this:

[6mm upcut]
G1 F3600 Z6

Several times on larger projects, continuing the next day, while having programed and problem solved on other projects,
I wonder if had used a 4mm or 6mm milling bit, so this little feature will help to not ruining a partly milled project

Siemenc, 2021-07-02:
Good idea! I think the lowest Z-point of the file could be good to have there as well for double checking, as written in this previous issue:
https://github.com/fellesverkstedet/Bark-beetle-parametric-toolpaths/issues/6#issuecomment-370234296

JensDyvik, 2021-07-02:
I agree too. It should be a quite easy fix. We just need to double check if g-code and shopbot code work in the same way for commenting out a line

uomou, 2021-07-05:
and to finish it, we could add material choice too:

--// hard wood - 6mm upcut 2 - z max -12.6 //--