fellesverkstedet/Bark-beetle-parametric-toolpaths #81: kill calc

uomou, 2021-11-24:
I wonder if there is a way to stop the calculation proces.
Say you have a full sheet with nested parts, sometimes it takes like 10 minutes until the calculation is doen.
You spot a mistake while bb is calculating, and I cannot aboard the calculation.

Thanks for help

Alex

JensDyvik, 2021-11-24:
I am not sure. Sometime hitting the escape button can stop a calculation process sooner in grasshopper. Have you done any research on generally stopping a calculation process in grasshopper?

uomou, 2021-11-24:
hitting escape doesent do much, but I just found this:

The new component Data Dam is pretty useful for keeping large definitions under control. A few Data Dam at key locations can turn a potentially BSODing large calculation into several smaller managable ones, and for calculations that does not change often you can have the data buffered at the dam (by disabling the stuff before it) to save CPU time. I'm liking this component a lot atm

JensDyvik, 2021-11-29:
My experience with the data dam component is that it is a bit finicky. I am also worried that the user interface will become more complicated to learn and use if you have to toggle a data dam component or more.

Maybe the best solution for now is to speed up general calculation time?

mikeFandA, 2021-12-01:
If you´re open to other plugins you could look at human, it has various
interface components that help with this kind of thing. Another option is
to setup up a small scripting component to do the same as data dam but
using a button press, then locate the buttons for the various stages with a
note to say apply changes to path, to block, to material etc. I haven't
dipped into the structure of your code for quite a while but it's possible
that you could split each function into several nodes that are calculated
separately before the final calculation (if you haven't already done so),
with some previews at each stages to verify correct data entry or results
to keep things user friendly and avoid heavy calculations for now reason.

*Michael Arnett*

BArch MArch

Coordinador de Robotica y Diseño Digital

Universidad Autonoma de Chile

Sede Temuco

+56 45 2895298

Avda. Rudecindo Ortega 01302, Temuco